efabless/mpw_precheck #3: License check not suitable

mithro, 2020-10-30:
(Plus, the current functionality is wrong.)

The tool should basically check for compliance with the stuff described at https://opensource.google/docs/releasing/preparing/ with an extra requirement to have SPDX identifiers in files.

There should be a list of approved licenses, it is more than just Apache 2.0 

A quick summary of the checks;
 - The top level repository has a LICENSE file (and the license is of an approved type). 
 - All third_party material should be under a third_party directory and have a license identifier. https://opensource.google/docs/releasing/preparing/#third-party-components
 - All documentation should use inclusive language https://opensource.google/docs/releasing/preparing/#inclusive
 - Repository should have a README file
https://opensource.google/docs/releasing/preparing/#include-a-readme-file
 - **All** text files should have a copyright header (and appropriate SPDX identifier). https://opensource.google/docs/releasing/preparing/#license-headers

You probably want to go look at Google's license checking tool at https://github.com/google/licenseclassifier and should go consider using an existing tool like FSF's reuse -- https://git.fsfe.org/reuse/tool 

Also checkout https://reuse.software/

agorararmard, 2020-10-30:
> * All documentation should use inclusive language https://opensource.google/docs/releasing/preparing/#inclusive
> 
> * Repository should have a README file
>   https://opensource.google/docs/releasing/preparing/#include-a-readme-file

@mithro: I will take out these two points into a separate issue because a different person is responsible for the License check, and these points are related to the documentation checks. (just to make tracking responsibility easier)

SaraEfabless, 2022-11-03:
Hi Mithro, 

Do you have any update on this? I need to either assign a subject matter expert to it or close the ticket. appreciate your help with it. 

Sara.