efabless/mpw_precheck #4: Documentation check not suitable

agorararmard, 2020-10-30:
(This is moved from #3 for ease of traceability)

It should check the following:
- All documentation should use inclusive language https://opensource.google/docs/releasing/preparing/#inclusive
- Repository should have a README file
        https://opensource.google/docs/releasing/preparing/#include-a-readme-file

agorararmard, 2020-11-06:
@mithro: Does that #7 resolve this?

mithro, 2020-11-08:
Could you create some tests of example repositories which should pass and fail these documentation checks? That way it doesn't matter how the implementation is written, only that the examples look right to me.

agorararmard, 2020-11-09:
@mithro: Briefly, until we prepare full repos of caravel test cases, what the documentation checks do currently is the following:
1. check for a top level (in the main page) README file of any txt file documentation format.
2. Reads all documentation files in all project sub-directories along with the main README file and checks for the use of the following words and their derivatives: slave, whitelist, blacklist. 

And that's about it for the documentation checks. 

Do you think we should force the inclusion of some sections with specific naming, i.e.: `How to Run` and so?

mithro, 2020-11-10:
@agorararmard - As we only have two checks, I think it is pretty simple to create some tests which should cover the failing and passing use cases?

mithro, 2020-11-10:
I also don't think we need to be any more strict about documentation.

agorararmard, 2020-11-10:
> @agorararmard - As we only have two checks, I think it is pretty simple to create some tests which should cover the failing and passing use cases?

@mithro: The thing is the doc checks are part of the fuzzy checks (not a separate script) but a function within a script. However, my reply above is the result of running some local tests with some tweaks in the requirements of the other checks (the non-doc checks).

We should be able to have full caravel test cases by the end of the week. (Hopefully)

agorararmard, 2020-11-10:
> I also don't think we need to be any more strict about documentation.

Roger that.

mithro, 2020-11-10:
@agorararmard Each check should be tested independently?

agorararmard, 2020-11-12:
@mithro: It was tested independently after disabling the rest of the checks (which is not the default behavior). Anyhow, what I did is that I tested on the OpenLANE repository without any changes and with adding some derivates of (slave, blacklist, whitelist) and seeing the result. I also tried deleting the main README.md file.

Each check was tested independently before the integration.

mithro, 2020-11-12:
@agorararmard -- Manual testing is testing that does not exist. Please set up testing so it can be run by any developer and by the CI system.

DavidRLindley, 2024-12-10:
I'm closing this ticket due to age and staleness.

closed 2024-12-10 by DavidRLindley