efabless/mpw_precheck #16: Problem with unicode characters

thesourcerer8, 2020-11-19:
I have put my name into the info.yaml:
[...]
  owner: "Philipp Gühring"
[...]

On the website, it seems to have worked fine, but when i run the following command inside docker:
python3 open_mpw_prechecker.py --target_path /caravel-stdcelllib-stdcells
I get this errors:

{{PROGRESS}} Executing Step 2 of 4: Checking YAML description.
'ascii' codec can't decode byte 0xc3 in position 317: ordinal not in range(128)
{{FAIL}} YAML file not valid in target path, please check the README.md for more info on the structure
TEST FAILED AT STEP 2

I guess that the docker container might need some Unicode support packages installed?

agorararmard, 2020-11-19:
I blieve this fixes this issue https://github.com/efabless/open_mpw_precheck/commit/fc8a7d1d6c73d5aac996883d3c0f6bab42ad44c4?

thesourcerer8, 2020-11-19:
Yes, looks good! Thanks!

closed 2020-11-19 by thesourcerer8