efabless/mpw_precheck #19: Spice Netlist Checking Incorrect

yrrapt, 2020-11-30:
The logic of the checker specifies that both the `top_level_netlist` and `user_level_netlist` be the same type (verilog or spice).

But there appears to be no `top_level_netlist` in spice format available and therefore the tool fails when trying to use a spice `user_level_netlist`.

agorararmard, 2020-11-30:
@yrrapt: ideally, we'll provide a spice netlist as well as an elaborated verilog netlist for the caravel chip in the caravel repository. In the meantime, you can run the extraction separately using [this script](https://github.com/efabless/caravel/blob/master/scripts/ext-gds-sky130A.sh) from the caravel repo on the caravel.gds. However, just to let you know, the power and signal routing on the top level are still a WIP in the caravel repo. If you're using spice you should point to these files in the `info.yaml` instead of the verilog files.

yrrapt, 2020-12-01:
@agorararmard Thanks for the info and instructions.
I have been running the extraction but it's 24+ hours with no signs of stopping. I don't have the most powerful computer though.
What is that field used for? Is it LVS or something else?

agorararmard, 2020-12-02:
@yrrapt: this is used to execute some consistency checks (kind of a fuzzy LVS check). However, I wouldn't rush into running extractions on caravel until at least power and signal routing are out and done and perhaps by then, we'd be providing a spice file in the caravel repo.

yrrapt, 2020-12-02:
@agorararmard No problem, I'll wait for you to complete your work.
Thanks the help.

a-omla, 2022-02-08:
The  consistency check today takes into consideration the nature of the project (analog/digital) and runs checks on the appropriate netlist. If issues still show up please open a new issue @yrrapt. Thank you 🙏