efabless/mpw_precheck #121: GPL-3.0 License failed precheck, it identifies as AGPL-3.0 (which it isn't)

wspitts2, 2022-03-17:
We would like to use GPL-3.0 License but it fails the precheck saying that AGPL-3.0 is prohibited. We used the following GPL-3.0 license file (we obviously called it LICENSE):
https://github.com/efabless/mpw_precheck/blob/mpw-5c/checks/license_check/_licenses/_approved_licenses/GPL-3.0.txt

Ran mpw_precheck and got the message: "A prohibited LICENSE (AGPL-3.0) was found" 

We compared it to the AGPL-3.0 License here: and the GPL-3.0 and AGPL-3.0 files are different in the license_check.
https://github.com/efabless/mpw_precheck/blob/mpw-5c/checks/license_check/_licenses/_prohibited_licenses/AGPL-3.0.txt

There is no AGPL license in the project .

The Apache 2.0 license does pass precheck so we know we are running it correctly. 

Thanks in advance.

chrische-xx, 2022-05-27:
I can confirm that this is still an issue. When i remove the AGPL and SSPL licenses from the precheck tool license folder it will work fine.

DavidRLindley, 2024-12-10:
I'm closing this ticket due to age and staleness.

closed 2024-12-10 by DavidRLindley