efabless/mpw_precheck #141: m4.pd.1d density check

msaligane, 2022-09-12:
Hi!

We are trying to close our submission (precheck is clean) however, the tapeout step (metal fill) is failing and gives the following error:
[image]

Unfortunately, I can only check the design and run metal fill on the open galaxy environment which takes a lot of time to run BTW(maybe an enhancement idea to let people run this locally).

This is our submission: https://platform.efabless.com/projects/1324

msaligane, 2022-09-12:
FYI, we tried to fix it by inserting m4 manually and then we were told actually we have to reduce metal fill density on m4.

msaligane, 2022-09-12:
Updating after [discussing](https://open-source-silicon.slack.com/archives/C01E06TUSC9/p1662999889103499) with @RTimothyEdwards on the open-source-silicon.dev slack workspace.

Quoting @RTimothyEdwards:

> Magic reports that metal4 density (metal density, not clear area) within the user project itself is 6.9%.
> ```
> % cif cover MET4
> Cell Area = 10329984000000 CIF units^2
> Layer Bounding Area = 9888485739600 CIF units^2
> Layer Total Area = 713447009075 CIF units^2
> Coverage in cell = 6.9%
> ```
> This might require getting @jeffdi  to manually override the precheck, while an issue is raised to root out and fix the problem, which (given the existing evidence) I conclude is either an error in klayout or in the density check definition.

RTimothyEdwards, 2022-09-13:
This looks like the issue:

(File https://github.com/efabless/mpw_precheck/blob/main/checks/drc_checks/klayout/met_min_ca_density.lydrc)

```
   m4_wildcard = "71/0-4,6-43,45-*"
   via4_wildcard = "71/44"
   m4fill_wildcard = "51/28"
```

This treats all purposes 0 to 4, 6 to 43, and 45 and up as "kinds of metal" that add to the density count.  In that range is purpose 10, which is fill block.  The first line above should be written

```
   m4_wildcard = "71/0-4,6-9,11-43,45-*"
```

and this needs to be done for *all* metal layers starting with local interconnect up to and including metal 5, since they are all similarly constructed wrong.

msaligane, 2022-09-13:
Thanks @RTimothyEdwards 
Should be fixed by: #142