efabless/mpw_precheck #151: XOR check failing

jeffdi, 2022-11-02:


jainsoumil2, 2022-11-25:
Hi, 
I was running the precheck locally by pulling [mpw_precheck commit `05fbcdaa43583f438826e9be2e2a258647515cea`](https://github.com/efabless/mpw_precheck/tree/05fbcdaa43583f438826e9be2e2a258647515cea) and was failing the XOR-check. I attached the precheck log and the xor check log files. 
@d-m-bailey suggested the problem could be due to the magic file begin updated - `Loading "/Volumes/export/isn/soumil/efabless/mpw_precheck/checks/xor_check/erase_box.tcl" from command line.
"sky130(vendor)" is not one of the CIF input styles Magic knows.`

[precheck.log](https://github.com/efabless/mpw_precheck/files/10087830/precheck.log)
[xor_check.log](https://github.com/efabless/mpw_precheck/files/10087831/xor_check.log)

RTimothyEdwards, 2022-11-25:
I don't think that would be an issue.  It will not cause magic to exit;  it will just return to the default input style, which is the one that is needed.

DavidRLindley, 2024-12-04:
Closing due to lack of response and the issue hasn't happened in the last four shuttles.

closed 2024-12-04 by DavidRLindley