efabless/mpw_precheck #161: sky130B_mr.drc is missing

m-usama-z, 2022-11-22:
The "Klayout FEOL" check fails when using sky130B PDK because "mpw_precheck/checks/tech-files/sky130B_mr.drc" is missing. Kindly resolve.

xobs, 2022-12-27:
To add on to this (since I experienced the same issue):

Running `make run-precheck` fails `Klayout FEOL`, `Klayout BEOL`, and `Klayout Offgrid`. The error reported for all of these is:

```
ERROR: Unable to open file: /si/work/caravel-gf180-pio/deps/precheck/checks/tech-files/sky130B_mr.drc (errno=2)
```

This occurs if you follow the steps at https://caravel-user-project.readthedocs.io/en/latest/#running-mpw-precheck-locally -- namely run `make precheck` followed by `make run-precheck`.

A simple symlink with `ln -s sky130A_mr.drc sky130B_mr.drc` inside `checks/tech-files/` allows it to pass those tests.

marwaneltoukhy, 2023-10-16:
This has been resolved.

closed 2023-10-16 by marwaneltoukhy