efabless/mpw_precheck #172: Traceback with gfmpw-0b prechecks

proppy, 2022-11-30:
```
{{STEP UPDATE}} Executing Check 3 of 4: Klayout BEOL
Traceback (most recent call last):
  File "mpw_precheck.py", line 141, in <module>
    default_content='_default_content')
  File "mpw_precheck.py", line 102, in main
    run_precheck_sequence(precheck_config=precheck_config, project_config=project_config)
  File "mpw_precheck.py", line 68, in run_precheck_sequence
    results[check.__surname__] = check.run()
  File "/home/runner/mpw_precheck/check_manager/__init__.py", line 156, in run
    self.klayout_cmd_extra_args)
  File "/home/runner/mpw_precheck/checks/drc_checks/klayout/klayout_gds_drc_check.py", line 20, in klayout_gds_drc_check
    with open(report_file_path) as klayout_xml_report:
FileNotFoundError: [Errno 2] No such file or directory: '/home/runner/work/tiny_user_project/tiny_user_project/precheck_results/30_NOV_2022___15_55_12/outputs/reports/klayout_beol_check.xml'
```

See https://github.com/proppy/tiny_user_project/actions/runs/3584751075/jobs/6032164785

proppy, 2022-12-01:
Maybe related to https://github.com/efabless/globalfoundries-pdk-libs-gf180mcu_fd_pr/pull/5

proppy, 2022-12-02:
I think this is due to:
https://github.com/KLayout/klayout/issues/1189
and fixed by:
https://github.com/KLayout/klayout/pull/1193

whutddk, 2022-12-05:
I have the same problem. 
https://platform.efabless.com/shuttles/GFMPW-0/project/1634
It can pass locally with 64GB memory. But failed online.
Can these project bypass the online precheck manually?

gatecat, 2022-12-05:
I have the same problem with both of these projects:
https://platform.efabless.com/projects/1505
https://platform.efabless.com/projects/1589

A reduced size version of the latter does pass online fine:
https://platform.efabless.com/projects/1613

gatecat, 2022-12-05:
Bumping klayout on the platform seems like the best solution here imo

egorxe, 2022-12-05:
Same here https://platform.efabless.com/projects/1637 . Precheck passes locally with latest KLayout but OOMs on platform during FEOL CO.6a.

AbdulMoizSheikh1, 2022-12-05:
Hi every one, I am having same issue for my projects.
https://platform.efabless.com/projects/1664 and https://platform.efabless.com/projects/1713
The FEOL is failing and after inspecting logs I am certain that this due to not running complete tests on the gds as the memory fills up and test fails. I have tried it with with 60 GB local machine but logs show same problem. 
Is the memory use-age related to design size too??
Also is there any way to reduce the memory useage in FEOL stage as the memory use-age is too much and this makes designing on not very high end systems quite impossible.

jeffdi, 2022-12-06:
Please go ahead and submit your project if you are able to successfully complete a precheck offline running locally.  

Do submit a precheck job on the platform and also include a comment in the messages for your shuttle submission that your precheck passes successfully locally.  

We will include it in the selection process and run the precheck and tapeout manually for your project.  Thanks.