efabless/mpw_precheck #178: Documentation check tries to check binary files

xobs, 2022-12-27:
Running `make run-precheck` in my project directory as described at https://caravel-user-project.readthedocs.io/en/latest/#running-mpw-precheck-locally nearly passes. The `Klayout FEOL`, `Klayout BEOL`, and `Klayout Offgrid` tests fail due to #161, but a simple symlink solves those.

There are, however, many many errors related to `SPDX COMPLIANCE FILE UNICODE DECODE EXCEPTION` errors. The process appears to be trying to decode binary files as UTF-8, then complains when it can't do that:

<details><summary>precheck.log summary</summary>
<p>

```
2022-12-27 10:14:31 - [INFO] - An approved LICENSE (Apache-2.0) was found in /si/work/caravel-gf180-pio.
2022-12-27 10:14:31 - [INFO] - {{SUBMODULES LICENSE CHECK PASSED}} No prohibited LICENSE file(s) was found in project submodules
2022-12-27 10:14:31 - [ERROR] - SPDX COMPLIANCE FILE UNICODE DECODE EXCEPTION in (/si/work/caravel-gf180-pio/.manual-flow.sh.swp): 'utf-8' codec can't decode byte 0xa7 in position 16: invalid start byte
2022-12-27 10:14:31 - [ERROR] - SPDX COMPLIANCE FILE UNICODE DECODE EXCEPTION in (/si/work/caravel-gf180-pio/wb_pio_test.elf): 'utf-8' codec can't decode byte 0xf3 in position 18: invalid continuation byte
2022-12-27 10:14:31 - [ERROR] - SPDX COMPLIANCE FILE UNICODE DECODE EXCEPTION in (/si/work/caravel-gf180-pio/openlane/wb_pio/runs/sky130B-2022-12-27_17_06/issue_reproducible/results/routing/wb_pio.odb): 'utf-8' codec can't decode byte 0xb9 in position 16: invalid start byte
2022-12-27 10:14:33 - [ERROR] - SPDX COMPLIANCE FILE UNICODE DECODE EXCEPTION in (/si/work/caravel-gf180-pio/openlane/wb_pio/runs/sky130B-2022-12-27_17_06/results/cts/wb_pio.odb): 'utf-8' codec can't decode byte 0xb9 in position 16: invalid start byte
2022-12-27 10:14:33 - [ERROR] - SPDX COMPLIANCE FILE UNICODE DECODE EXCEPTION in (/si/work/caravel-gf180-pio/openlane/wb_pio/runs/sky130B-2022-12-27_17_06/results/floorplan/wb_pio.odb): 'utf-8' codec can't decode byte 0xb9 in position 16: invalid start byte
2022-12-27 10:14:33 - [ERROR] - SPDX COMPLIANCE FILE UNICODE DECODE EXCEPTION in (/si/work/caravel-gf180-pio/openlane/wb_pio/runs/sky130B-2022-12-27_17_06/results/placement/wb_pio.odb): 'utf-8' codec can't decode byte 0xb9 in position 16: invalid start byte
2022-12-27 10:14:33 - [ERROR] - SPDX COMPLIANCE FILE UNICODE DECODE EXCEPTION in (/si/work/caravel-gf180-pio/openlane/wb_pio/runs/sky130B-2022-12-27_17_06/results/routing/wb_pio.odb): 'utf-8' codec can't decode byte 0xb9 in position 16: invalid start byte
2022-12-27 10:14:36 - [ERROR] - SPDX COMPLIANCE FILE UNICODE DECODE EXCEPTION in (/si/work/caravel-gf180-pio/openlane/wb_pio/runs/sky130B-2022-12-27_17_06/tmp/12-wb_pio.odb): 'utf-8' codec can't decode byte 0xb9 in position 16: invalid start byte
2022-12-27 10:14:36 - [ERROR] - SPDX COMPLIANCE FILE UNICODE DECODE EXCEPTION in (/si/work/caravel-gf180-pio/openlane/wb_pio/runs/sky130B-2022-12-27_17_06/tmp/cts/11-wb_pio.resized.odb): 'utf-8' codec can't decode byte 0xb9 in position 16: invalid start byte
...
2022-12-27 10:15:49 - [ERROR] - SPDX COMPLIANCE FILE UNICODE DECODE EXCEPTION in (/si/work/caravel-gf180-pio/deps/pdks/skywater-pdk/env/conda/envs/skywater-pdk-scripts/compiler_compat/ld): 'utf-8' codec can't decode byte 0xa5 in position 25: invalid start byte
2022-12-27 10:15:50 - [ERROR] - SPDX COMPLIANCE FILE UNICODE DECODE EXCEPTION in (/si/work/caravel-gf180-pio/deps/pdks/skywater-pdk/env/conda/x86_64-conda-linux-gnu/bin/ld): 'utf-8' codec can't decode byte 0xf8 in position 96: invalid start byte
2022-12-27 10:15:50 - [ERROR] - SPDX COMPLIANCE FILE UNICODE DECODE EXCEPTION in (/si/work/caravel-gf180-pio/deps/pdks/skywater-pdk/env/conda/compiler_compat/ld): 'utf-8' codec can't decode byte 0xf8 in position 96: invalid start byte
2022-12-27 10:16:24 - [WARNING] - {{SPDX COMPLIANCE CHECK FAILED}} Found 60286 non-compliant file(s) with the SPDX Standard.
2022-12-27 10:16:24 - [INFO] - SPDX COMPLIANCE: NON-COMPLIANT FILE(S) PREVIEW: ['/si/work/caravel-gf180-pio/manual-flow-sky130.sh', '/si/work/caravel-gf180-pio/synth.tcl', '/si/work/caravel-gf180-pio/activate-caravel.sh', '/si/work/caravel-gf180-pio/manual-flow.sh', '/si/work/caravel-gf180-pio/openlane/wb_pio/config.json', '/si/work/caravel-gf180-pio/openlane/wb_pio/runs/sky130B-2022-12-27_17_06/config_in.tcl', '/si/work/caravel-gf180-pio/openlane/wb_pio/runs/sky130B-2022-12-27_17_06/runtime.yaml', '/si/work/caravel-gf180-pio/openlane/wb_pio/runs/sky130B-2022-12-27_17_06/config.tcl', '/si/work/caravel-gf180-pio/openlane/wb_pio/runs/sky130B-2022-12-27_17_06/issue_reproducible/run.sh', '/si/work/caravel-gf180-pio/openlane/wb_pio/runs/sky130B-2022-12-27_17_06/issue_reproducible/env.lldb', '/si/work/caravel-gf180-pio/openlane/wb_pio/runs/sky130B-2022-12-27_17_06/issue_reproducible/run.tcl', '/si/work/caravel-gf180-pio/openlane/wb_pio/runs/sky130B-2022-12-27_17_06/issue_reproducible/env.gdb', '/si/work/caravel-gf180-pio/openlane/wb_pio/runs/sky130B-2022-12-27_17_06/issue_reproducible/openlane/scripts/openroad/sta_multi_corner.tcl', '/si/work/caravel-gf180-pio/openlane/wb_pio/runs/sky130B-2022-12-27_17_06/issue_reproducible/openlane/scripts/openroad/common/set_global_connections.tcl', '/si/work/caravel-gf180-pio/openlane/wb_pio/runs/sky130B-2022-12-27_17_06/issue_reproducible/openlane/scripts/openroad/common/io.tcl']
2022-12-27 10:16:24 - [INFO] - For the full SPDX compliance report check: /si/work/caravel-gf180-pio/precheck_results/27_DEC_2022___10_14_20/logs/spdx_compliance_report.log
2022-12-27 10:16:24 - [INFO] - {{STEP UPDATE}} Executing Check 2 of 14: Makefile
2022-12-27 10:16:24 - [INFO] - {{MAKEFILE CHECK PASSED}} Makefile valid.
2022-12-27 10:16:24 - [INFO] - {{STEP UPDATE}} Executing Check 3 of 14: Default
2022-12-27 10:16:25 - [INFO] - {{README DEFAULT CHECK PASSED}} Project 'README.md' was modified and is not identical to the default 'README.md'
2022-12-27 10:16:25 - [INFO] - {{CONTENT DEFAULT CHECK PASSED}} Project 'gds' was modified and is not identical to the default 'gds'
2022-12-27 10:16:25 - [INFO] - {{STEP UPDATE}} Executing Check 4 of 14: Documentation
2022-12-27 10:16:25 - [WARNING] - The documentation file (/si/work/caravel-gf180-pio/deps/pdks/skywater-pdk/env/conda/pkgs/conda-4.12.0-py39h06a4308_0/info/test/tests/data/env_metadata/py36-osx-whl/lib/python3.6/site-packages/urllib3-1.23.dist-info/DESCRIPTION.rst) contains the non-inclusive word: whitelist
2022-12-27 10:16:27 - [WARNING] - The documentation file (/si/work/caravel-gf180-pio/deps/pdks/open_pdks/sources/precheck_sky130/debug_precheck.md) contains the non-inclusive word: blacklist
2022-12-27 10:16:27 - [WARNING] - The documentation file (/si/work/caravel-gf180-pio/deps/precheck/debug_precheck.md) contains the non-inclusive word: blacklist
2022-12-27 10:16:27 - [WARNING] - {{DOCUMENTATION CHECK FAILED}} Project documentation is not appropriate.
2022-12-27 10:16:27 - [INFO] - {{STEP UPDATE}} Executing Check 5 of 14: Consistency
```
</p>

Additionally, it complains that files such as my config.json are non-compliant, despite the fact that the JSON file format does not support comments:

```
2022-12-27 10:16:24 - [INFO] - SPDX COMPLIANCE: NON-COMPLIANT FILE(S) PREVIEW: ['/si/work/caravel-gf180-pio/manual-flow-sky130.sh', '/si/work/caravel-gf180-pio/synth.tcl', '/si/work/caravel-gf180-pio/activate-caravel.sh', '/si/work/caravel-gf180-pio/manual-flow.sh', '/si/work/caravel-gf180-pio/openlane/wb_pio/config.json', '/si/work/caravel-gf180-pio/openlane/wb_pio/runs/sky130B-2022-12-27_17_06/config_in.tcl', '/si/work/caravel-gf180-pio/openlane/wb_pio/runs/sky130B-2022-12-27_17_06/runtime.yaml', '/si/work/caravel-gf180-pio/openlane/wb_pio/runs/sky130B-2022-12-27_17_06/config.tcl', '/si/work/caravel-gf180-pio/openlane/wb_pio/runs/sky130B-2022-12-27_17_06/issue_reproducible/run.sh', '/si/work/caravel-gf180-pio/openlane/wb_pio/runs/sky130B-2022-12-27_17_06/issue_reproducible/env.lldb', '/si/work/caravel-gf180-pio/openlane/wb_pio/runs/sky130B-2022-12-27_17_06/issue_reproducible/run.tcl', '/si/work/caravel-gf180-pio/openlane/wb_pio/runs/sky130B-2022-12-27_17_06/issue_reproducible/env.gdb', '/si/work/caravel-gf180-pio/openlane/wb_pio/runs/sky130B-2022-12-27_17_06/issue_reproducible/openlane/scripts/openroad/sta_multi_corner.tcl', '/si/work/caravel-gf180-pio/openlane/wb_pio/runs/sky130B-2022-12-27_17_06/issue_reproducible/openlane/scripts/openroad/common/set_global_connections.tcl', '/si/work/caravel-gf180-pio/openlane/wb_pio/runs/sky130B-2022-12-27_17_06/issue_reproducible/openlane/scripts/openroad/common/io.tcl']
```

mithro, 2022-12-30:
FYI - @jeffdi / @mkkassem

marwaneltoukhy, 2023-06-07:
License issue fixed here: https://github.com/efabless/mpw_precheck/pull/195