efabless/mpw_precheck #194: mpw-9a: spdx compliance check flags files in default caravel_user_project

d-m-bailey, 2023-03-30:
Running precheck on the default `caravel_user_project` (after `make setup`) detects SPDX non-compliance in the following files and directories.

Files:
```
project/openlane/user_proj_example/no_synth.cells
project/openlane/user_proj_example/base_user_proj_example.sdc
project/openlane/user_proj_example/config.json
project/openlane/user_proj_example/drc_exclude.cells
project/openlane/user_project_wrapper/base_user_project_wrapper.sdc
project/openlane/user_project_wrapper/config.json
project/openlane/user_project_wrapper/fixed_dont_change/no_synth.cells
project/openlane/user_project_wrapper/fixed_dont_change/drc_exclude.cells
project/lib/user_project_wrapper.lib
project/lib/user_proj_example.lib
project/verilog/includes/includes.gl+sdf.caravel_user_project
project/verilog/includes/includes.gl.caravel_user_project
project/verilog/includes/includes.rtl.caravel_user_project
```

Directories:
```
venv
dependencies
signoff
mgmt_core_wrapper
```
Expectation:
There should be no compliance issues with the default data.

d-m-bailey, 2023-06-07:
@marwaneltoukhy 
Here's a full list of the SPDX exceptions for the mpw-9d default project.
[spdx_compliance_report.log](https://github.com/efabless/mpw_precheck/files/11671533/spdx_compliance_report.log)
Eliminating these directories
```
dependencies
runs
signoff
venv
```
leaves
```
/home/user/mpw-9/cup-mpw9d/sdf/user_proj_example.sdf
/home/user/mpw-9/cup-mpw9d/sdf/user_project_wrapper.sdf
/home/user/mpw-9/cup-mpw9d/sdf/multicorner/nom/user_project_wrapper.tt.sdf
/home/user/mpw-9/cup-mpw9d/sdf/multicorner/nom/user_project_wrapper.ff.sdf
/home/user/mpw-9/cup-mpw9d/sdf/multicorner/nom/user_project_wrapper.ss.sdf
/home/user/mpw-9/cup-mpw9d/sdf/multicorner/min/user_project_wrapper.tt.sdf
/home/user/mpw-9/cup-mpw9d/sdf/multicorner/min/user_project_wrapper.ff.sdf
/home/user/mpw-9/cup-mpw9d/sdf/multicorner/min/user_project_wrapper.ss.sdf
/home/user/mpw-9/cup-mpw9d/sdf/multicorner/max/user_project_wrapper.tt.sdf
/home/user/mpw-9/cup-mpw9d/sdf/multicorner/max/user_project_wrapper.ff.sdf
/home/user/mpw-9/cup-mpw9d/sdf/multicorner/max/user_project_wrapper.ss.sdf
/home/user/mpw-9/cup-mpw9d/sdc/user_proj_example.sdc
/home/user/mpw-9/cup-mpw9d/sdc/user_project_wrapper.sdc
/home/user/mpw-9/cup-mpw9d/openlane/user_proj_example/no_synth.cells
/home/user/mpw-9/cup-mpw9d/openlane/user_proj_example/base_user_proj_example.sdc
/home/user/mpw-9/cup-mpw9d/openlane/user_proj_example/config.json
/home/user/mpw-9/cup-mpw9d/openlane/user_proj_example/drc_exclude.cells
/home/user/mpw-9/cup-mpw9d/openlane/user_project_wrapper/base_user_project_wrapper.sdc
/home/user/mpw-9/cup-mpw9d/openlane/user_project_wrapper/config.json
/home/user/mpw-9/cup-mpw9d/openlane/user_project_wrapper/fixed_dont_change/no_synth.cells
/home/user/mpw-9/cup-mpw9d/openlane/user_project_wrapper/fixed_dont_change/drc_exclude.cells
/home/user/mpw-9/cup-mpw9d/lib/user_project_wrapper.lib
/home/user/mpw-9/cup-mpw9d/lib/user_proj_example.lib
/home/user/mpw-9/cup-mpw9d/verilog/includes/includes.gl+sdf.caravel_user_project
/home/user/mpw-9/cup-mpw9d/verilog/includes/includes.gl.caravel_user_project
/home/user/mpw-9/cup-mpw9d/verilog/includes/includes.rtl.caravel_user_project
/home/user/mpw-9/cup-mpw9d/lvs/user_project_wrapper/sky130A.lvs_config.sh
/home/user/mpw-9/cup-mpw9d/spef/user_project_wrapper.spef
/home/user/mpw-9/cup-mpw9d/spef/user_proj_example.spef
/home/user/mpw-9/cup-mpw9d/spef/multicorner/user_project_wrapper.min.spef
/home/user/mpw-9/cup-mpw9d/spef/multicorner/user_project_wrapper.nom.spef
/home/user/mpw-9/cup-mpw9d/spef/multicorner/user_project_wrapper.max.spef
```
I think we can ignore the `.cells`, `.lib`, `.sdc`, `.sdf`, `.spef` suffixed files. Which leaves
```
/home/user/mpw-9/cup-mpw9d/openlane/user_proj_example/config.json
/home/user/mpw-9/cup-mpw9d/openlane/user_project_wrapper/config.json
/home/user/mpw-9/cup-mpw9d/verilog/includes/includes.gl+sdf.caravel_user_project
/home/user/mpw-9/cup-mpw9d/verilog/includes/includes.gl.caravel_user_project
/home/user/mpw-9/cup-mpw9d/verilog/includes/includes.rtl.caravel_user_project
/home/user/mpw-9/cup-mpw9d/lvs/user_project_wrapper/sky130A.lvs_config.sh
```
I think we can add the headers to the `includes` files and `lvs_config`, but `.json` files do not natively support comments. Do we exclude `.json` for now?

marwaneltoukhy, 2023-06-07:
Thank you for the comprehensive issue, I think we should exclude the `.json` files for now yes. I will work on the rest.

marwaneltoukhy, 2023-06-07:
fixed here: https://github.com/efabless/mpw_precheck/pull/195

PR #202 (2023-06-07): SPDX-compliance exceptions added.
Also added `lvs_results` directory to lvs output hierarchy.

Included `.json` files because they're not set up to include comments.

Should fix #194
closed 2023-06-22 by marwaneltoukhy