efabless/mpw_precheck #201: add precheck for connecting user_oeb

jeffdi, 2023-05-24:


jeffdi, 2023-05-24:
@RTimothyEdwards we need documentation for the gpio block.

RTimothyEdwards, 2023-05-25:
What form of documentation do you want, and where do you want it to go?  I have plenty of documentation of the GPIO from the original MPW-one documentation that needs only minor updates for the current version.

RTimothyEdwards, 2023-05-25:
Also note that the documentation for the GF version of the GPIO control block is different from that for the sky130 version, so we will need to make sure that we organize the documentation to make clear which version of the caravel chip the documentation corresponds to, and to make sure that we have proper documentation for both versions (and any future additional versions) of the caravel chip.

RTimothyEdwards, 2023-05-25:
FYI, the spreadsheet that I made recently for the SPI and wishbone memory maps of the GPIO is only correct for the sky130 version.  When I get the chance, I'll split off a version for GF and make the appropriate changes.  Of course, that's only part of the documentation, but it's a major component of it.

jeffdi, 2023-07-05:
test developed - needs integration with precheck @marwaneltoukhy.  

need to make extraction standalone.  this should support any user_project_wrapper GDS, regardless of the design flow used to create it.

closed 2023-08-13 by marwaneltoukhy