efabless/mpw_precheck #225: 2309: Track extra_be_checks integration

d-m-bailey, 2023-09-06:
Having a clear method of tracking what version of `extra_be_checks` is included in `mpw_precheck` is needed.

Current commit of `extra_be_checks` is `a60939f2844da06a8907d04c3368d17de5b1db84` and contains samples of `lvs_config` files used in sky130 mpw-3 and sky130 mpw-4.

marwaneltoukhy, 2023-09-10:
I agree, but `extra_be_checks` has files that doesn't belong in precheck, like the lvs configurations for other projects. Precheck should only contain scripts that runs the checks. I have a couple of recommendations:
1. Remove `extra_be_checks` and create a fork from precheck, where you can store this info, and any updates to precheck scripts can be easily integrated
2. Create a CI that creates a PR whenever an update happens to `extra_be_checks` and only include the updates to scripts in this PR (this adds a layer of complexity that I'm not sure we want)

d-m-bailey, 2023-09-11:
@marwaneltoukhy Thanks for the comments.

What we definitely don't want is manual file editing every time there's an update.

`mpw_precheck` is probably always going to be a subset of `extra_be_checks`.
`mpw_precheck` currently is designed for `user_project_wrapper` or `user_analog_project_wrapper`, while `extra_be_checks` works at any cell level. Are there plans to do something with precheck for `openfame`?

Because, `extra_be_checks` is designed independent of any preconceived hierarchy, the suggestion of removing it, forking `mpw_precheck` and making changes there seems ill-advised to me. I don't the user should be required to download the `mpw_precheck` if they're not using openlane.

All updates to `extra_be_checks` are not critical to `mpw_precheck` because `mpw_precheck` only uses a subset of the information. I don't know that we want automated PR's created for every update.

I was looking into sub modules and that seems like a simple and clean way to track what version `mpw_precheck` is using without adding to or changing the `mpw_precheck` file structure. It looks like some devs don't like sub modules though.

What are your thoughts?

marwaneltoukhy, 2023-09-11:
@d-m-bailey 
`mpw_precheck` is used for any kind of project that is going to be submitted on the platform, so it is independent of OpenLane. Precheck is designed to test designs going on the shuttle to insure a smooth integration with caravel, and increase reliability of the design. 

I haven't received any requests to have precheck for `openframe`, but I assume that there's a plan for it.

We'll have to take the infrastructure's team input on submoduling the `extra_be_checks` repo, as precheck is not only a local tool, but it is also used on the platform.