efabless/mpw_precheck #230: add version check to precheck

jeffdi, 2023-09-14:


jeffdi, 2023-09-14:
Perhaps this should be a hard failure.

marwaneltoukhy, 2023-09-18:
@jeffdi I think this check might be better to be implemented in the user project's makefile, as the makefile is responsible for the user's environment, and is aware of which tag the user is currently using.

We could print a warning that the precheck version is incorrect and exit, and provide the user with the target they can run to update precheck (which is `make precheck`)

jeffdi, 2023-09-19:
@marwaneltoukhy i think that misses the objective.  we want precheck to check its version against the current available.  the point is not to lock in on a tag, but to enable us to roll out a new version and users will automatically be prompted to update the next time they run, regardless of the version of caravel_user_project they started their project with (which can be behind in many cases).

jeffdi, 2023-09-19:
there is never a case where a user should be using an old version of precheck.

jeffdi, 2024-01-24:
PR pending - still needs final review

jeffdi, 2024-02-28:
create new version of user flow with the proposed changes in the PR.

DavidRLindley, 2024-12-04:
This is implicated in the ChipIgnite CLI tool.  Marwan/Anton/DL to meet offline to determine path.