efabless/mpw_precheck #231: review drc rules and precheck steps currently implemented for GF180

marwaneltoukhy, 2023-09-24:


marwaneltoukhy, 2023-09-24:
Precheck missing checks:

1. LVS
2. OEB
3. Consistency
4. Defaults (makes sure that the readme and gds are different)
5. Documentation (makes sure the documentation is readable and doesn't contain non-inclusive words)
6. Pin label purpose overlapping drawing
7. zero area
8. magic DRC
9. Makefile (makes sure the makefile has the required targets `'check-caravel', 'check-pdk', 'check-precheck', 'clean', 'help', 'install', 'openlane', 'precheck', 'run-precheck', 'simenv', 'simlink', 'uninstall', 'update_caravel', 'verify'`)
10. manifest (makes sure the user is using the correct caravel)

marwaneltoukhy, 2023-09-24:
DRC rules list:
https://docs.google.com/spreadsheets/d/10AvjBIigrxH9b7_KaqSEuhQlGs3enxP4jqt_ax2O9Ek/edit?usp=sharing

jeffdi, 2023-09-27:
Target for precheck:

- OEB
- defaults
- documentation
- Makefile
- manifest

marwaneltoukhy, 2023-10-03:
For the OEB we need the tech files for `gf180mcuD`

marwaneltoukhy, 2023-10-10:
Timing profile on the example project https://docs.google.com/spreadsheets/d/1X5Z1ui3BesCzzAIfZDyxrgVrL7Cjmlfw02ltszCsruI/edit#gid=0

marwaneltoukhy, 2023-11-26:
Done here #244

closed 2023-11-26 by marwaneltoukhy