efabless/mpw_precheck #256: mpw-9: when tech file is missing, drc passes with 0 errors.

d-m-bailey, 2023-11-08:
Missing tech file `sky130A.tech` in `sky130B` pdk.

```
Magic 8.3 revision 420 - Compiled on Sat Nov  4 21:03:17 UTC 2023.
Starting magic under Tcl interpreter
Using the terminal as the console.
Using NULL graphics device.
Processing system .magicrc file
Sourcing design .magicrc for technology sky130A ...
2 Magic internal units = 1 Lambda
Could not find file '/root/eda/pdk/sky130B/libs.tech/magic/sky130A.tech' in any of these directories:
         . $CAD_ROOT/magic/sys $CAD_ROOT/magic/sys/current
Error parsing "/root/mpw_precheck/checks/tech-files/sky130A.magicrc": couldn't read file "/root/eda/pdk/sky130B/libs.tech/magic/sky130A.tcl": no such file or directory
Bad local startup file "/root/mpw_precheck/checks/tech-files/sky130A.magicrc", continuing without.
Scaled tech values by 2 / 1 to match internal grid scaling
Loading "/root/mpw_precheck/checks/drc_checks/magic/magic_drc_check.tcl" from command line.
Don't know how to read GDS-II:
Nothing in "cifinput" section of tech file.
[INFO]: Loading user_project_wrapper
Cell user_project_wrapper couldn't be read
No such file or directory
Creating new cell
"drc(full)" is not one of the DRC styles Magic knows.
The current style is "default".
The DRC styles are: default.
No errors found.
[INFO]: COUNT: 0
[INFO]: Should be divided by 3 or 4
[INFO]: DRC Checking DONE (/root/eda/caravel_user_project/precheck_results/08_NOV_2023___10_53_00/outputs/reports/magic_drc_check.drc.report)
[INFO]: Saving mag view with DRC errors(/root/eda/caravel_user_project/precheck_results/08_NOV_2023___10_53_00/outputs/user_project_wrapper.magic.drc.mag)
[INFO]: Saved
```

0 DRC errors reported.

d-m-bailey, 2023-11-08:
Deeper problem.
```
    magicrc_file_path = parent_directory.parent.parent / 'tech-files' / 'sky130A.magicrc'
```
`sky130A.magicrc` is hardcoded in `checks/drc_checks/magic/magic_gds_drc_check.py`

However, `sky130A.magicrc` has
```
tech load $PDKPATH/libs.tech/magic/sky130A.tech

# load device generator
source $PDKPATH/libs.tech/magic/sky130A.tcl

# load bind keys (optional)
# source $PDKPATH/libs.tech/magic/sky130A-BindKeys
```
These `sky130A`  files do not exist in the `sky130B` of `gf180mcuD` pdks.

There is a `tech-files/sky130B.tech` file but it is a symbolic link to `sky130A.tech`.

This is probably also a problem for `gf180mcu`.

RTimothyEdwards, 2023-11-09:
This could be handled with something like
```
if {[tech name] != "sky130A"} {exit 1}
```
in the script.

RTimothyEdwards, 2023-11-09:
But ultimately the problem is that there _is_ no sky130A.tech in the sky130B technology, and there shouldn't be.  The error is something like PDK is set to "sky130B" but somewhere "sky130A" has been hard-coded.

d-m-bailey, 2023-11-10:
Exactly. If the technology could not be read, I think the scripts would expect magic to fail, not generate a result that says no drc errors. Is that something that could easily be added to the batch version (`-dnull -noc`) of magic?

RTimothyEdwards, 2023-11-10:
@d-m-bailey :  `utils/main.c` line 1144 would be where to put it, coupled with the code in the same file at line 608---Calling magic in batch mode sets Tcl variable "batch_mode" which can be checked in order to do things specifically in batch mode only.

d-m-bailey, 2023-11-10:
@RTimothyEdwards you fixed the hardcoded file name in #229 but it was never merged.

marwaneltoukhy, 2023-11-26:
Merged

closed 2023-11-26 by marwaneltoukhy