efabless/mpw_precheck #266: create check in precheck & tapeout flow to verify only expected layers exist for the process stack

jeffdi, 2024-02-28:


marwaneltoukhy, 2024-03-03:
Blocked on the complete list of valid layers that can be taped out. I think the complete list is not open sourced.

DavidRLindley, 2024-04-10:
Marwan commented in the Slack channel `shuttle-development` that this was the most complete layer map he could find.

https://vault.efabless.com/foundry/sw-s130/-/blob/master/V2.1.305/LIBS/S130/S130.layermap

I compared the layers in this file to that of the S8 Calibre DRC deck.  For the vast majority the S130.layermap file is a superset.  The only omissions from the S130.layermap file are

1. tap.drawing  65/44 
2. hvi.drawing  75/20 
3. areaid.waffleWindow
4. prBoundary.boundary  235/250
5. prBoundary.drawing    235/252

This comparison is in 

https://docs.google.com/spreadsheets/d/17zUXO5DpSXaFgCOAFVSzJoFpuvrg0FoVtSt_DIqGK-E/edit#gid=0

I have no idea why tap and hvi are missing from `S130.layermap`.   They seem to be important layers.

I think with the addition of the tap and hvi the drawn layer list in S130.layermap is probably complete.  But, there are additional layers in our post-tapeout Caraval die (based on a sample size of one 2404 die and the first 8 slots of 2309) that are not in either of the layer maps.  The layers are the added waffles.  

| LAYER | DATATYPE | Name |
|: --- :|: --- :|: --- :|
|23 | 28 | cfom |
|28 | 28 | cp1m |
|34 | 28 | cmm3 |
|36 | 28 | cmm1 |
|41 | 28 | cmm2 |
|51 | 28 | cmm4 |
|56 | 28 | cli1m |
|59 | 28 | cmm5 |

Marwan, can you write a script to verify the layers in a GDS/OAS file against the list of drawing/created layers in the S130.layermap file (along with the tap, hvi, and the extra waffle layers)?  I'll run it on the 2309 die.

RTimothyEdwards, 2024-04-11:
@DavidRLindley :  TAP is missing because in the S8 definition, TAP is replaced by DIFF, which is more in line with most foundry process definitions (it is a TAP because of the coincident type of implant;  marking TAP as separate from DIFF is entirely redundant).  HVI was replaced by THKOX and 5V (the latter being an identifier layer only).  I thought that one of those had the same GDS numbers as HVI but was only a name change.

DavidRLindley, 2024-12-04:
This check is coded in a PR.  It is waiting on a layers file.  Marwan to send a PR link to DL, DL to push closed.