efabless/mpw_precheck #273: Consistency check fails for openframe project using vccd2/vssd2

urish, 2024-04-20:
> 2024-04-20 21:42:35 - [WARNING] - PORTS CHECK FAILED: openframe_project_wrapper ports do not match the golden wrapper ports. Mismatching ports are : ['vccd2', 'vssd2']

Is that the expected behavior?

smunaut, 2024-04-20:
Ok, to be a bit more clear, what we were trying is to use vccd2/vssd2 just as supplement power ingress to the PDN ring, so they are shorted to vccd1/vssd1.  We don't need different rails, but we could use a better PDN / ingress path ...

The way we attempted it was just to change the template and remove the vccd2/vssd2 pins and move their geometry to vccd1/vssd1.

RTimothyEdwards, 2024-04-21:
To be even more clear, I suggested (strongly) that domains should be tied together because one point of the openframe version was to work around dumb restrictions like not being able to merge power domains together.  It is absolutely ridiculous to try to drive a 15mm^2 chip core from a single pin, and even more ridiculous when there are at least three pins readily available to feed the power supply (I would actually recommend tying together vccd1, vccd2, and vccd, and possibly tossing the three vdda domains in there as well).

So, to whoever is responsible for this (I will add some likely candidates to the assignees list):  Figure out how to make it work.

RTimothyEdwards, 2024-04-22:
@urish @smunaut :  I had sort of assumed that something like what was done for https://github.com/efabless/caravel_openframe_project with the cells `vccd1_connection` and `vssd1_connection` would be extensible to add in connections to other domains.  That would, I expect, require that cells other than vccd1/vssd1 might need to split the power lines with metal resistors to prevent them from being reported as shorts.

smunaut, 2024-04-22:
We were planning to wire vccd/vssd in there as well, but the automatic script we have to wire up the rings to the power pins currently only supports left/right side and not top/bottom ( pins are on met3, rings on top/bottom are on met5 so it's a bit more complicated and I didn't have the time to work that out for this tapeout, so this will be for the next one ).

smunaut, 2024-04-23:
To keep things updated here, I did manage to have the rails internally shorted and still exported as independent pins name ( vccd1 / vccd2 ) externally. And this does make the pre-check happy. ( I didn't try further / tapeout check or anything ).

Unfortunately this requires some changes to openroad ( https://github.com/parallaxsw/OpenSTA/issues/27 ) because ATM when it exports netlists to verilog it doesn't support to have power pins connected together.  The SPICE extraction is fine ( different ports, shorted by a resistor which is what we want ), but the verilog netlist that it's compared against during LVS is written by OpenRoad ( `write_verilog -include_pwr_gnd` ) and this will have the two ports unconnected which obviously leads to LVS error.