efabless/mpw_precheck #284: Create illegal cell name character check in precheck and tapeout flows

DavidRLindley, 2024-07-10:


urish, 2024-09-06:
Looks like magic sometimes creates cells with "#". We've just hit this on Tiny Tapeout:

[image]

dlmiles, 2024-09-06:
@RTimothyEdwards 

Candidate offending cause could be auto-renaming during cell read-in by Magic VLSI tool

https://github.com/RTimothyEdwards/magic/blob/master/database/DBio.c#L2122 currently there is no config option to change suffix method renaming strategy to be compatible with this new precheck.

DavidRLindley, 2024-09-06:
Hi Uri,

If that is the only error, just edit the user_project_wrapper.gds to
replace the illegal character with something else.  A unique number per
instance maybe. I think I used "__" in your last tapeout. Let me know once
you've done so and I'll push it into tapeout.  Hopefully we'll have a fix
before the next shuttle.

David

urish, 2024-09-06:
Thanks David! It was @algofoogle's project, and he swiftly pushed a fix.

urish, 2024-09-06:
I believe so - we'll retry tonight after the tt08 submission deadline

DavidRLindley, 2024-09-06:
Alright!  Let me know if you need anything.

DavidRLindley, 2024-09-25:
done

closed 2024-09-25 by DavidRLindley